Sirui-Xu/InterMimic #3: What's the difference between dof_pose and body_pose?

kunkun0w0, 2025-04-06:
Dear authors,

Thanks for releasing this amazing work! I am a little confused about the structure of the `hoi_data` in `_load_motion`.
 https://github.com/Sirui-Xu/InterMimic/blob/5e83a631c3eb53bca52f3d6ac3109e9527e9da99/intermimic/env/tasks/intermimic.py#L147-L169

Could you explain the meaning of the different dimensions in `loaded_dict['hoi_data']`, especially the difference between `dof_pos` and `body_pose`?

Thanks for your help!

Sirui-Xu, 2025-04-06:
Great questions! We'll add comments to this section of the code and improve its readability in short. Are there any specific variables that confused you? I'd be happy to clarify them right away.

kunkun0w0, 2025-04-07:
Thanks for replying! I am confused about the `dof_pos` and `root_pos`. 

For `dof_pos`, should I directly use the [full_pose](https://github.com/vchoutas/smplx/blob/1265df7ba545e8b00f72e7c557c766e15c71632f/smplx/body_models.py#L1301) in SMPLX body or the original pose without adding the mean pose? And which rotation representation should I use?

For `root_pos`, should I use the translation of the SMPLX body or the position of the root joint?

Thanks for your help!

Sirui-Xu, 2025-04-07:
Hi @kunkun0w0,

You are welcome! To briefly clarify, **dof_pos** is represented using an exponential map for local rotation, whereas **root_pos** corresponds to the root joint and is defined in a different coordinate system within Isaac Gym.

For further details, please refer to the following resources:
- [PHC convert_data_smpl.py](https://github.com/ZhengyiLuo/PHC/blob/master/scripts/data_process/convert_data_smpl.py)
- [SMPLSim](https://github.com/ZhengyiLuo/SMPLSim)

I used these code bases for constructing the data. I hope this helps, and feel free to reach out if you have any further questions.

Best regards.

closed 2025-04-08 by Sirui-Xu