Sirui-Xu/InterMimic #4: hand penetration during picking up the table?

erwincoumans, 2025-04-07:
When running the sh scripts/test.sh, the hands are penetrating the table during pickup, see video
https://github.com/user-attachments/assets/c6fbf0f1-44a9-4dc9-b125-308f8e4eeeb1
Is the script running an animation? Is there a special fixed constraint during contact/pickup of the table?

Sirui-Xu, 2025-04-07:
Hi @erwincoumans, thanks for raising this excellent question. As mentioned in Section H of our paper, the policy can sometimes result in unnatural object support. In some cases, there is slight interpenetration between the human and the object, which can be mitigated by setting a higher depenetration velocity, applying a stronger contact energy to discourage large contact forces. 

While full penetration, as the example you shared, is rare, it did occur in our original InterMimic model. We found that incorporating object affordance, rather than relying solely on the interaction vector in InterMimic (human joint -> object surface), in both the representation and the mimicking reward largely resolves the issue. We plan to release this improvement in the future.

erwincoumans, 2025-04-07:
Thanks for the information,  good to know.

closed 2025-04-07 by erwincoumans