Sirui-Xu/InterMimic #6: Reducing GPU Memory Usage When Running scripts/test.sh

Chpppy, 2025-04-11:
Hi Sirui Xu,

First of all, thank you for sharing your excellent work on InterMimic—your implementation is clear, well‑structured, and a fantastic contribution to the community. I’ve been following the repository closely, and I really appreciate the effort you put into making the code easy to understand and extend.

I’m trying to reproduce your results on my own machine, but I’ve run into a CUDA out‑of‑memory issue when running the test script. Here are the details of my setup:

- **GPU:** NVIDIA RTX 3070 Ti with 8 GB VRAM  
- **OS:** Ubuntu 20.04  
- **Command:** `sh scripts/test.sh`  
- **Modifications tried:** I reduced `--nums_envs` to 2, but the error persists.  

Below is the relevant error output:

```
[Error] [carb.gym.plugin] Gym cuda error: out of memory: …/GymPhysX.cpp:1721
[Error] [carb.gym.plugin] Gym cuda error: invalid resource handle: …/GymPhysX.cpp:6137
[Error] [carb.gym.plugin] Gym cuda error: out of memory: …/GymPhysXCuda.cu:991
…  
RuntimeError: CUDA error: an illegal memory access was encountered  
…  
```

I’m wondering if there are additional parameters or configuration options I could tweak to reduce the GPU memory footprint. For example:

- Would lowering the physics substep count or reducing the simulation batch size help?  
- Are there any flags in `sim_params` or elsewhere to limit the number of concurrent environments or simplify the observation tensors?  
- Would compiling with different CUDA settings (e.g., enabling device‑side assertions) provide more insight or reduce memory usage?

Could you please advise on the best way to adjust the code or parameters to avoid these out‑of‑memory errors? Any guidance would be greatly appreciated!

Thank you again for your time and for all the great work.

Sirui-Xu, 2025-04-11:
Hi @Chpppy, thank you so much for your kind words. Could you please pull the latest version of the code, where I’ve applied some optimizations, and let me know if it resolves your issue?

I also appreciate you raising this issue—it motivates me to further organize and improve the codebase. If the problem persists, my understanding is that most simulation parameters mainly affect efficiency rather than actually releasing GPU memory. You might consider reducing the number of points in the point cloud, though this could impact the performance of the policy.

Chpppy, 2025-04-15:
Thank you for your prompt reply and continued support. I pulled the latest code from the repository and re‑ran `scripts/test.sh`, but the out‑of‑memory issue persisted. Next, I tried the G1 example, and it ran successfully on my GPU, so I compared the two configuration files. I noticed that in `intermimic/data/cfg/omomo.yaml`, the line

```yaml
sim:
  substeps: 2
  physx:
    ...
    # max_gpu_contact_pairs: 34603008 # 8*1024*1024
```

is active, whereas in `intermimic/data/cfg/omomo_g1_29dof_with_hand.yaml` this parameter is commented out. After commenting out `max_gpu_contact_pairs` in `omomo.yaml`, the test script executed without errors.

Could you please explain what the `max_gpu_contact_pairs` setting controls in PhysX, and what the consequences might be of disabling or commenting out this option? I want to make sure I’m not inadvertently affecting simulation fidelity or performance.

Thank you again for your help and for all the great work on this project.

Sirui-Xu, 2025-04-15:
Hi, thanks for your insightful observation!

The max_gpu_contact_pairs parameter sets the maximum number of contact pairs for which GPU memory is **reserved**. During inference, it's unlikely that the simulation will generate a very large number of contact pairs, so you can safely comment out this line if needed. However, during training, significant penetration from policy exploration may result in an excessive number of contact pairs, and if the limit isn’t set high enough, the simulation might produce errors (like NaN values).

We will soon release an alternative training setting along with new checkpoints that offer more accurate simulations for training, addressing the potential for large contact pairs even for training, though this will come with some efficiency trade-offs.

Chpppy, 2025-04-16:
Thanks for your detailed explanation! It‘s very helpful to me.

Thanks again!

closed 2025-04-16 by Chpppy