Sirui-Xu/InterMimic #9: Training for g1 CANNOT be reproduced

YuenFuiLau, 2025-04-23:
Thanks for your contribution. May I ask for some clarification of training on G1 robot?

I tried to train g1 based on the training script and here is the cmd that I used for g1 training:

`python intermimic/run.py --task InterMimicG1 --cfg_env intermimic/data/cfg/omomo_g1_29dof_with_hand.yaml --cfg_train intermimic/data/cfg/train/rlg/omomo_g1_29dof_with_hand.yaml --headless --output checkpoints`

I tried for 8000 epochs but the result seems weird:

[image]

Sirui-Xu, 2025-04-23:
Hi,

We haven't organized that part of the code yet—currently, only inference is supported for G1. If you try to train directly, the reward from the SMPL-X parent class will be used.

Best

Sirui-Xu, 2025-04-23:
We'll be releasing that part of the code early next month, stay tuned!

YuenFuiLau, 2025-04-23:
Glad to know about it. Thanks

mulplue, 2025-05-19:
Hi @Sirui-Xu , will G1 code be released recently? Really looking forward to try it out🥰

Sirui-Xu, 2025-05-21:
@mulplue Hello! just resumed work on open-sourcing this project this week. I am afraid I cannot tell a concrete date yet, the plan is to organize the PSI/student-policy training first, followed by the G1 training soon after, actively in the following weeks. Everything will be available before CVPR in mid-June.

Thanks for your patience!

mulplue, 2025-05-22:
> [@mulplue](https://github.com/mulplue) Hello! just resumed work on open-sourcing this project this week. I am afraid I cannot tell a concrete date yet, the plan is to organize the PSI/student-policy [...]

Thanks for your kind reply and great efforts in open-sourcing! I'll keep following the latest updates.