Sirui-Xu/InterMimic #11: A little help asking for the OMOMO format about trans2joint

Kairobo, 2025-04-24:
Why is there a trans2joint in the original OMOMO repo? Like:

root_trans = curr_global_root_jpos + curr_trans2joint # T X 3
mesh_jnts, mesh_verts, mesh_faces =
run_smplx_model(root_trans[None].cuda(), curr_local_rot_aa_rep[None].cuda(),
betas.cuda(), [gender], self.ds.bm_dict, return_joints24=True)

I know it's not your responsibility to answer this, but I'd be glad if you could spare some time to help me. Thanks!

Sirui-Xu, 2025-04-24:
Hi @Kairobo ,

You are welcome!! Please excuse any inaccuracies—I may not recall the detail of the OMOMO codebase. I think the reason should be that the SMPL translation (`root_trans`) does not align exactly with the pelvis joint position (`curr_global_root_jpos`) but with an offset. You’ll see a similar adjustment in my other projects:

- **InterDiff**:  
  https://github.com/Sirui-Xu/InterDiff/blob/d584b1f117684f9fa3b76b69c06b7b9ffde905c2/interdiff/data/dataset_smpl.py#L127  
- **InterAct**:  
  https://github.com/wzyabcas/InterAct/blob/main/process/process_omomo.py

Kairobo, 2025-04-25:
Very Cool. Thank you!

closed 2025-04-25 by Kairobo