Sirui-Xu/InterMimic #16: About the reproduction of teacher policy training

Liam-lyr, 2025-05-09:
Hi, thank you for the excellent work!

I am encountering some issues while attempting to reproduce the training of sub2.pth (the teacher policy of subject2 in OMOMO). I would greatly appreciate your assistance in resolving some of my confusions:

1. **About the training config**: I noticed discrepancies between the reward weights in omomo.yaml and those provided in the appendix of the paper. Was the omomo.yaml in the repository used for training the released sub2.pth? Could you provide the training configuration for this teacher policy?
2. **About the training code**：Are the currently open-sourced checkpoints trained using the present training code? I noticed in #7 that you mentioned PSI has not yet been released.
3. **About the training data**: It appears that some motion clips originally belonging to OMOMO are missing from the retargeted subject 2 motions in the current repository (e.g. sub2_largetable_006, sub2_woodchair_009, ...). May I ask that were the results in the paper and the open-sourced checkpoints trained on all reference motions of the respective subjects in the retargeted OMOMO dataset?

Thank you very much for the assistance!

Sirui-Xu, 2025-05-09:
Hi,

Thank you very much for your feedback. Currently, the provided code mainly supports the basic training pipeline and does not yet include some features available in the configuration used for training the released checkpoint, such as slightly different weights, PSI, the full data for training, and several other small functionalities. These features are somewhat complex to organize, but I'll make sure to prepare and release them sometime next week.

Feel free to let me know more details about the issue you encountered while using the current version to train the model. I'd be happy to help.

Best

Sirui-Xu, 2025-05-09:
Yes, that's true for sub2.pth and sub8.pth (G1). sub2_high_fid.pth is trained using the current version.