Sirui-Xu/InterMimic #19: Input data format

hlinhn, 2025-06-06:
Thank you for the great work! I have a question regarding the format of the given OMOMO data, a little similar to issue #3. In the following I assume that the decoding of the given .pt file is done in `env/tasks/intermimic.py`, correct me if that is wrong.
- how are `contact_human` generated? (indices 331 to 331+52). At first I thought it would be similar to `contact_obj` (just on/off), but there seem to be 3 types of values (-1, 0, 1), can you explain what they mean?
- I understood that the 52 DOFs are from the robot definition `omomo.xml`. How do you convert SMPL body to this representation and vice-versa? 

Is there a script to convert from SMPL-(X/H) and object position/rotation to this representation?  

Thank you so much!

hlinhn, 2025-06-08:
awesome, thank you!

Sirui-Xu, 2025-06-25:
Hi @hlinhn, you're right — for OMOMO, since the hand positioning is incorrect, we don't necessarily rely on the hand-related data for representation and reward formulation. As a result, it should have only a minor impact on policy performance. We've fixed several issues in the data and provided the updated version as OMOMO_new.

closed 2025-07-04 by hlinhn