Sirui-Xu/InterMimic #26: Data issues when migrating to isaaclab

tianshuwu, 2025-07-17:
Dear authors,

Thanks for releasing this amazing work! I tried to follow your work in isaaclab, but I encountered some problems when applying the reference data directly: When i set the joints of robot to the motion data you offered with the isaaclab api"write_joint_state_to_sim", it didn't perform as expected. 

Is there any difference in the expected input when setting the robot's dof state between isaaclab and isaacgym? Or is there any invisible preprocessing in your code? 

Looking forward to your answer!

What I tried:
1.  I noticed that the real position of the quaternion was opposite in isaacgym and isaaclab, so swapped the positions so the pos and rot of the root were set correctly.
2. I aligned the joint names, so the dof should be in the right order.

[image]

Sirui-Xu, 2025-08-01:
Hi,

Please feel free to correct me if I’m mistaken, my understanding of Isaac Lab is quite limited. I think you've already addressed the key differences:

    Isaac Gym stores quaternions in xyzw format, whereas Isaac Lab uses wxyz.

    Isaac Gym traverses the kinematic tree in a depth-first manner. Isaac Lab may use a different order, but it seems you’ve already adjusted for that.

    The USD importer in Isaac Lab might bake in coordinate transformations that aren't present in my XML file in Gym.

Hope this helps!

Best regards

tianshuwu, 2025-08-06:
Thanks for your reply！