Sirui-Xu/InterMimic #31: A Bug in Computer Humanoid Observation?

Kairobo, 2025-09-14:
_ref_body_pos = self.extract_data_component('body_pos', obs=ref_obs).view(ref_obs.shape[0], -1, 3)[:, key_body_ids, :]
        _body_pos = body_pos[:, key_body_ids, :]

        diff_global_body_pos = _ref_body_pos - _body_pos
        diff_local_body_pos_flat = torch_utils.quat_rotate(flat_heading_rot_2, diff_global_body_pos.view(-1, 3)).view(-1, len_keypos * 3)

        local_ref_body_pos = **_body_pos** - root_pos.unsqueeze(1)  # preserves the body position
        local_ref_body_pos = torch_utils.quat_rotate(flat_heading_rot_2, local_ref_body_pos.view(-1, 3)).view(-1, len_keypos * 3)

the **_body_pos** should be _ref_body_pos?

Sirui-Xu, 2025-09-16:
Hi,

Good catch, the original intent was indeed to include the local reference motion here, so using _ref_body_pos is correct. Since the observation already incorporates the reference motion through the difference term, this likely won’t lead to significant performance differences.

Apologies for the delay in releasing the new architecture and yogamat data -- I was blocked by some other things recently. I’ll finish wrapping it up shortly and share it here this week.

Best

Kairobo, 2025-09-17:
No problem! Thank you for the efforts of releasing them!

closed 2025-09-17 by Kairobo