Sirui-Xu/InterMimic #33: Running the pipeline in cuda:1

Kairobo, 2025-09-17:
I try to use a different device (cuda:1 instead of cuda:0) to run the script:
I try to pass the argument rl_device = 1
or 
    args = get_args(force_test=True)
    # for gpu control
    if True:
        choose_gpu_id = 1
        args.compute_device_id = choose_gpu_id
        args.device_id = choose_gpu_id # for simulation
        args.rl_device = f'cuda:{choose_gpu_id}' # for policy
        args.sim_device = f'cuda:{choose_gpu_id}'
but all would trigger the following error:
[Error] [carb.gym.plugin] Gym cuda error: an illegal memory access was encountered: ../../../source/plugins/carb/gym/impl/Gym/GymPhysXCuda.cu: 948
[Error] [carb.gym.plugin] Gym cuda error: an illegal memory access was encountered: ../../../source/plugins/carb/gym/impl/Gym/GymPhysXCuda.cu: 1001

  File "/home/amax/kaijia/projects/InterMimic/intermimic/env/tasks/humanoid.py", line 189, in _reset_env_tensors
    self.reset_buf[env_ids] = 0
RuntimeError: CUDA error: an illegal memory access was encountered
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

Do you have a way to use cuda:1?

Sirui-Xu, 2025-09-17:
Hello,

Are you using multiple GPUs for training? In my experience, Horovod in this codebase can be tricky to work with, so I’d recommend using Torch Distributed instead, it should be much easier to set up and manage.

The warning you’re seeing is most likely related to GPU out-of-memory issues. Do you have any logs of GPU memory usage (e.g., from a wandb curve)? If so, does it show memory use close to 100%?

Best

Kairobo, 2025-09-18:
No. I'm just trying to get two runs on separate gpus. The GPU memory should be enough for the running. I guess there are some hidden tensors get used in the other gpu.

Kairobo, 2025-09-18:
Thank you very much! Very good advice.